cncjs/cncjs-shopfloor-tablet #1: Screenshots

MitchBradley, 2017-10-17:
[image]

MitchBradley, 2017-10-18:
Revised layout and names for axis setting buttons
[image]

MitchBradley, 2017-12-14:
Added 2D visualization

[image]

qiwalker, 2019-01-25:
hi mr. @MitchBradley  

how to get list of gcode file sir
i can find anyting in drop down menu gcode file
im sory for my bad english

Thanks Rizki

karoria, 2019-03-06:
@MitchBradly
Sir, I am unable to get the program list using shopfloor tablet. I doubt if i missed any process to configure watch directory. I have manually created a directory named "watch" under home/pi
Kindly guide me.

MitchBradley, 2019-03-06:
I added instructions for setting up a watch directory to https://github.com/cncjs/cncjs-shopfloor-tablet/blob/master/README.md

MitchBradley, 2020-03-10:
[image]
[image]
[image]
[image]
Firewall
[image]

Firewall advanced
[image]

Firewall off

[image]

Security highlighting inbound rules

[image]

New Rule

[image]

Port

[image]

TCP 8000
[image]

Allow

[image]

Domain, Private

[image]

Description

[image]

Success

[image]

MitchBradley, 2020-11-24:
[image]