cncjs/cncjs-shopfloor-tablet #2: MDI boxes not interpreting g code correctly

ENGR-mike, 2017-12-24:
Using the latest release of shopfloor tablet.  Decided to try the MDI boxes and discovered the lower box intrprets coordinates incorrectly.  Upper MDI box interprets y80 as 80mm in the y direction.  But the lower MDI box intrpreted y-80 as something much larger and it hit the limit switch.  I tried y-8 just for the heck of it and it moved 80mm in the -y direction.  I tried x-20 in the lower MDI box and it tried to move what seemed like 200mm in th -x direction, but the limit switch stopped it.  I tried x-2 and it moved 20mm in the -x direction.  The upper MDI box does not seem to have any problems as far as I can tell.

MitchBradley, 2017-12-25:
As I have said, it is really important to give complete information.  What exactly did you type?  Just 'y80'?That would be ambiguous without knowing the previous context of which motion command (G0, G1, G2, etc) was executed last, and whether the modal state is absolute (G90) or relative (G91.)

I tried "G0 Y80" in the upper MDI followed by "G0 Y-80" in the lower MDI - starting from a reset with the Y coordinate initially at 0 and the default G90 absolute mode.  The result was as expected - it moved 80 mm in the +Y direction to Y = 80.000, then it moved 160 mm in the -Y direction to Y = -80.000.

[image]

The console window in the full UI showed that the commands that were typed into the MDI boxes were transmitted faithfully to Grbl:

[image]

It's important to know that the tablet UI does not "interpret" the MDI commands; rather it transmits them verbatim to Grbl for interpretation.  Of course it is conceivable that, perhaps, the tablet UI might somehow be inadvertently garbling a command, but I see no evidence of that from the test shown above.

ENGR-mike, 2018-01-03:
I have not been able to reproduce the problem since first reporting it.  If I am able to in the future, I will post additional details on the problem.