cncjs/cncjs-shopfloor-tablet #3: Path Error on execution

nickm324, 2018-03-07:
Getting the following error when I try and run the cmd to start. Also how do you run this in the background, how to make it work with pm2?

 **error: uncaughtException: "path" is not defined:** null date=Wed Mar 07 2018 16:05:06 G                    MT-0600 (CST), pid=7529, uid=1000, gid=1000, cwd=/usr/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nod                    e, version=v8.9.4, argv=[/usr/bin/node, /usr/bin/cnc, -m, /tablet:/home/pi/cncjs-shopfloor-tablet/src], rss=1291                    91936, heapTotal=102518784, heapUsed=86225476, external=357078, loadavg=[2.05908203125, 1.70556640625, 1.3447265                    625], uptime=1154, trace=[column=19, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=new Serial                    Connection, line=2018, method=null, native=false, column=27, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index                    .js, function=new GrblController, line=17181, method=null, native=false, column=34, file=/usr/lib/node_modules/c                    ncjs/dist/cnc/app/index.js, function=null, line=16817, method=null, native=false, column=13, file=events.js, fun                    ction=emitTwo, line=126, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=214, me                    thod=emit, native=false, column=12, file=/usr/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js, funct                    ion=null, line=503, method=null, native=false, column=7, file=internal/process/next_tick.js, function=_combinedT                    ickCallback, line=131, method=null, native=false, column=9, file=internal/process/next_tick.js, function=process                    ._tickCallback, line=180, method=_tickCallback, native=false], stack=[TypeError: "path" is not defined: null,                         at new SerialConnection (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:2018:19),     at new GrblController                     (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:17181:27),     at Socket.<anonymous> (/usr/lib/node_modules/                    cncjs/dist/cnc/app/index.js:16817:34),     at emitTwo (events.js:126:13),     at Socket.emit (events.js:214:7),                         at /usr/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js:503:12,     at _combinedTickCallback (in                    ternal/process/next_tick.js:131:7),     at process._tickCallback (internal/process/next_tick.js:180:9)]

ruswerner, 2018-04-16:
I'm also getting this error. To start it with pm2 you need to run this:

    pm2 start $(which cncjs) -- --port 8000 -m /tablet:/home/pi/cncjs-shopfloor-tablet/src

ruswerner, 2018-04-16:
@MitchBradley I'm running latest versions of CNCjs on Node 8.x LTS on Raspberry PI 3/debian stretch. Cncjs is running in pm2 but this error also happens when started directly using the command in your readme. Any ideas on what is going on here? or alternative instructions to get it working?

Cheers!

MitchBradley, 2018-04-16:
I get that error too, but have not seen any bad results from it.

I tried several times to debug it but always got lost in a twisty maze of little passages.  It seems that there is a timer-activated callback in the Socket library that is trying to instantiate a GrblController object automagically, without supplying a path to the serial device.  Perhaps this is happening prior to the time that the GrblController is opened properly with a valid path.  But why that would be happening, I cannot figure out.  Control flow in cncjs is byzantine, and the ultra-dynamic nature of modern Javascript and Node.js makes it very difficult to trace what is happening.

ruswerner, 2018-04-16:
@MitchBradley thanks for quick reply! Yeah I suppose you are right. I was having a problem where it wasn't prompting me to choose the Arduino port before the tablet interface loaded, and then I was stuck. I thought it was related to this, but it's not. :-)

You can probably close this issue then. Thanks!