cncjs/cncjs-shopfloor-tablet #5: No serial port connection

bluespider42, 2018-08-08:
After starting with the command:  `$ cnc -m /tablet:/home/pi/cncjs-shopfloor-tablet/src`  
And navigating to the page:  `http://<ip_address>:8000/tablet/'  
It redirects to `http://<ip_address>:8000/tablet/#/connection` however there is no serial port in the ports list.  
If I go straight to `http://<ip_address>:8000/tablet/#/axes` then it says "NoConnect" in the top left, if i click any of the buttons then I get the message `serial port "" not available`.  

CNCjs works fine and the serial port shows up.

MitchBradley, 2018-08-08:
My bad.  In order to avoid accidentally selecting the wrong serial port, I restricted the code to offer only the serial port that is connected to my machine.  That change is easy to undo by commenting out two lines.  I just made a branch with the change

https://github.com/cncjs/cncjs-shopfloor-tablet/compare/MitchBradley-patch-all-serial-ports

If you want to see just the changes, here is the commit

https://github.com/cncjs/cncjs-shopfloor-tablet/commit/a22e50ee91f8e5d2df2474eeebc4bf3bfbb72b05

bluespider42, 2018-08-09:
Thanks, 
I noticed that when I use https://github.com/cncjs/cncjs-pendant-tinyweb as well and select the serial port there then this works.

bluespider42, 2018-08-21:
The changes suggested work fine, thanks

closed 2018-08-21 by bluespider42