cncjs/cncjs-shopfloor-tablet #7: Error: listen EADDRINUSE

tigert10, 2018-09-03:
Please note that I am not an experienced programmer.  However, if I am following your steps correctly I am getting a consistent error.
I have unzipped and copied the folder to: /home/pi/cncjs-tablet
Then I PuTTY into my Raspberry Pi and run the command:
$ cnc -m /tablet:/home/pi/cncjs-tablet/src

It gives me the following error:
pi@cncjs:~ $ cnc -m /tablet:/home/pi/cncjs-tablet/src
2018-09-03T17:09:23.203Z - info: init Loading configuration from "/home/pi/.cncrc"
2018-09-03T17:09:23.307Z - info: init Mounting a directory "/home/pi/cncjs-tablet/src" to serve requests starting with /tablet
2018-09-03T17:09:27.185Z - error: init { Error: listen EADDRINUSE 0.0.0.0:8000
    at Object._errnoException (util.js:992:11)
    at _exceptionWithHostPort (util.js:1014:20)
    at Server.setupListenHandle [as _listen2] (net.js:1355:14)
    at listenInCluster (net.js:1396:12)
    at doListen (net.js:1505:7)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }
Error: { Error: listen EADDRINUSE 0.0.0.0:8000
    at Object._errnoException (util.js:992:11)
    at _exceptionWithHostPort (util.js:1014:20)
    at Server.setupListenHandle [as _listen2] (net.js:1355:14)
    at listenInCluster (net.js:1396:12)
    at doListen (net.js:1505:7)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000 }

MitchBradley, 2018-09-03:
EADDRINUSE means that CNCjs is already running , so the attempt to run it again fails because the other pre-existing process has exclusive use of the network communication port. The other prices was probably started automatically if you followed the instructions that tell you to set up pm2.
There are two ways to fix it. One is to manually kill the other CNCjs and then rerun the command line to start the tablet version. The other is to modify the parameters of the autostart command to include the tablet mount arguments.
I can't be more detailed at the moment because I am doing this on my phone. Later when I can sit down in front of a keyboard perhaps I can give more details

tigert10, 2018-09-03:
Thanks for the quick reply. That makes sense. I will see if I can find and
edit the autostart command.
Thanks again.

MitchBradley, 2018-09-04:
Closing the ticket based on the positive response.  Feel free to comment again or re-open if necessary.

closed 2018-09-04 by MitchBradley

tigert10, 2018-10-19:
Hi Mitch, not sure where I should ask this question because it is not
really an issue but a question.  I have the shop floor version of cncjs
running well now and use it often. However there is one feature from the
desktop version that I miss in the shop floor version and that is the macro
command section.  I have created an android app that generates a macro for
3 axis homing based on bit diameter.  The app generates the g-code and
copies it to the clip board.  If I paste that code into the macro section
of the full site of cncjs it runs great and finds a perfect 3 axis zero.  I
thought I could do the same thing with the shop floor version just pasting
the code into the MDI section.  However, it doesn't run the entire series
of commands (separated by ;) it only runs the first one.  Is there anyway
to paste about 20 lines of g-code into the MDI section at once and have it
run it one line at a time? Or, is there anyway I could add the macro
section from the desktop version?  Or do you have any other suggestions on
how I might be able to accomplish this?  Please let me know if I can
clarify in any way.  Thanks in advance...love the work you have done on
this!

karoria, 2019-01-17:
@MitchBradley I get from this thread discussion that at a time only one UI can be accessed. Either full UI or Tablet UI or Tiny UI, right? If so, is there any workaround if I want to enable all the UIs by default at startup of raspberry pi server and connect any UI as per my choice on different devices?
I am sorry if my question is very basic to you, but I am stuck with the same error EADDRINUSE and I want to use 2 different UI on different devices. Please help.

wesgentry, 2019-01-31:
wondering the same as above.. Would like to run both UI.. big one on my main workstation across the shop and shopfloor on a tablet mounted on my machine..