cncjs/cncjs-shopfloor-tablet #8: Nothing displayed in the port drop down

eciramella, 2018-11-04:
I'm sure it's something I'm doing, but with two pendants loaded (the tiny one for my phone and this one for my rpi touchscreen), the shopfloor one does not show anything in the ports dropdown.

I've successfully connected via the phone pendant and the normal cncjs interface, but for whatever reason, this one shows nothing in the ports drop down.

I'm happy to dig/poke/hack at this, but haven't found anything useful yet.

MitchBradley, 2018-11-04:
I hardwired the serial port list so it shows only the one connected to my TingG controller.  That keeps me from making silly mistakes that cause problems in the shop.  Here is a commit that undoes that change, restoring all the serial ports to the menu.

https://github.com/cncjs/cncjs-shopfloor-tablet/pull/6/commits/a22e50ee91f8e5d2df2474eeebc4bf3bfbb72b05

MitchBradley, 2019-02-03:
You need the server part of cncjs to run the cncjs-shopfloor-tablet user interface.  The full cncjs user interface is not necessary.  The server can run on a small computer like a Raspberry Pi.  In my setup, I have an Arduino Due running g2core firmware, in the same enclosure as a Raspberry Pi Zero W running the cncjs server, connected via wifi to an Android tablet that displays the cncjs-shopfloor-tablet user interface.

qiwalker, 2019-02-20:
thanks Mr. @MitchBradley

one more question
how to get list of gcode file sir
i can find anyting in drop down menu gcode file
im sory for my bad english

Regards Rizki

closed 2019-06-05 by MitchBradley