cncjs/cncjs-shopfloor-tablet #10: Unable to get program list

karoria, 2019-03-06:
@MitchBradly
Sir, I am unable to get the program list using shopfloor tablet. I doubt if i missed any process to configure watch directory. I have manually created a directory named "watch" under home/pi
Kindly guide me.

MitchBradley, 2019-03-06:
You must add a "watchDirectory" line inside the .cncrc file, inside the braces, like this:

```
{
    "watchDirectory": "/home/pi/watch",

```
followed by whatever was already in that file

karoria, 2019-03-06:
Got it. Thanks a lot. One more question.. Can I change default feed values of buttons (0.001 to 5) on shopfloor pendant simply by editing html code or it requires some workaround in js? As I work in mm, those values are too small to work.

MitchBradley, 2019-03-07:
Yes, it's just HTML.  In src/index.html there are a bunch of sections like:

```
                    <button
                        type="button"
                        class="btn btn-default set-distance"
                        title="Jog by 0.5"
                        onclick="cnc.setDistance('0.5')"
                    >
                        0.5
                    </button>
```

Just change the numbers per your needs.

karoria, 2019-03-07:
Many thanks :)

closed 2019-03-11 by karoria