cncjs/cncjs-shopfloor-tablet #12: Problem with 2D visualizer

karoria, 2019-04-08:
Hi @MitchBradley.
I have installed your new version of shopfloor tablet on my Raspberry Pi and found good changes as you have mentioned.
My concern is, now I can't see 2D visualization after update. This also occured in past, at that time you suggested to install a different branch named "patch all ports". Can you explain me why this happens so that I can remove the root cause of the problem. I doubt on the default port it opens while it connects.
Secondly, I was not able to click on feedrate override (new feature in tablet) while machine was running. All those buttons stays disabled. Is that having any relation to the above problem or you have just put those buttons but not programmed yet for functionality?
Let me inform you that I am planning for building an economic industrial VMC machine with shopfloor pendant as its default HMI.

MitchBradley, 2019-04-08:
I have this missing 2D visualization problem sometimes but when I try to debug it the problem goes away, i.e. the visualization reappears. That makes it very difficult to debug! My best guess is that something is getting cached and the process of trying to debug it refreshes the cache. I don't know what your browsing environment looks like so I can't give exact details, but if you are using Chrome, you might try turning on Developer Tools and reloading the page.  That tends to make the problem disappear, and then it is okay for some time.  Another thing to try is to browse to the page on a different machine.
The feedrate override is turned off at the moment because I have not yet finished the "plumbing" to make it work.  I'm working on it.  I'm not sure what to do about the spindle override at the moment, because g2core does not actually do anything with spindle override.  The scaffolding to handle it is present, but the functions that do the actual work - spindle_start_override() and spindle_end_override() - just return immediately.  I haven't researched how controllers other than g2core handle spindle control.

karoria, 2019-04-09:
Hi Mitch. Thanks for the answer. I want to draw your attention that while I start full console of cncjs, select the gcode file and 3d visualization once appears, then after if i close that full console and open shopfloor console, the visualization works. 
Second thing, once you get through feedrate override it is almost ok. As for spindle, we have analog ways which is much better solution then using touch. However, finally an analog tuner like Pot for feedrate override would be the best. 
My present spindle is 0-10v analog input. I have a signal from g2core converted to 0-10v analog and other is a pot which you can manually tune to 0-10v range. I have put a changeover switch between this two and this has made it manageable with spindle override (when override required, i put my changeover to pot mode and by adjusting pot, i get desired rpm).

MitchBradley, 2019-04-09:
Thanks for the clue about visualization.  I'll keep looking.
I have an override pot on my spindle too.
I hope to make some progress on the feedrate thing today.

MitchBradley, 2019-04-10:
I am running into all sorts of problems with the feedrate  control thing.  It's pretty clear that the g2core support for on-the-fly feedrate manipulation is a work-in-progress, and there is little or no progress at the moment. Several things:

- The documentation about the experimental feedrate feature https://github.com/synthetos/g2/wiki/Feed-Rate-Override does not match what is currently implemented.  The doc says the property name is "mfo", but the current g2core code uses "fro".  The cncjs code was written around the "mfo" formulation, but just changing that use "fro" doesn't work because
- The enabler property "froe" cannot be set with the usual method of "{froe:1}".  The command is accepted but does not actually change the underlying variable.
- However, you can cause "froe" to be set by issuing an "M50 Pn" command.  But that doesn't help much because, when you are running a gcode job, any attempts to then change the feedrate are queued up and only issued after the job is finished.
- The feedrate override doc page suggests that the feature is intended to be used with the "dual USB" setup, which makes sense because that setup would let you issue control commands over a separate channel that bypasses the gcode queue.  But that setup apparently had other problems and seems to have been abandoned.
- The g2core team has been relatively unresponsive to issues lately, making me suspect that they are spending their time on other things that pay better.  So I'm not holding my breath waiting for feedrate improvements.
- The difficulty of getting this to work correctly is inherent to the system architecture of "low end microcontroller receiving a gcode stream over a serial line with multiple queues in various places".  There are probably workarounds, but they are tricky and thus bug-prone.  GRBL finesses it by having four single-character commands for -10%, -1%, +1%, and +10%.  A single-character command doesn't require queue management.

The feedhold feature is not important enough to me to spend more time on it, so for now it is on indefinite hold.  I shall remove the artifacts from the shopfloor tablet UI, except for a display of the current velocity.  If you need to override the feedrate before a run, you can use MDI to issue an M50 command.  "M50 P2" is double the programmed feedrate, "M50 P0.1" is 1/10 rate, "M50" cancels  the override.  There are "M50.1 Pn" commands for overriding the rapid rate but g2core currently accepts but ignores them.

karoria, 2019-04-11:
So Sad. Anyway, thanks a lot to you for putting all your efforts going deep into the problem and investing your precious time.

MitchBradley, 2019-06-05:
I just pushed a change that fixed some problems with the 2D visualizer.  I don't know if it will fix your problem though.