cncjs/cncjs-shopfloor-tablet #13: Interface hangs while making work coordinates Zero with g2core edge

karoria, 2019-04-13:
CNCjs shopfloor tablet problem:
Hi Mitch, I am using latest shopfloor tablet. I noticed, while making X=0 or Y=0 or Z=0, the interface hangs with latest edge build of g2core.
CNCjs universal problem:
When limit switches (I use NC swithces on both side of each axis) operate, the console shows limit hit but doesn't generate alarm as in the case of master branch. Even if the limit hits, it lets the machine run normally. Anything you suggest trying in settings_default.h file?

MitchBradley, 2019-04-13:
Have a look at this https://github.com/synthetos/g2/issues/367 .  Down near the bottom there is a PR that fixes it.  As with all good code mods, it works by deleting one line :-).
The fact that the g2core authors have not acted on it is is evidence of the hypothesis that g2core is not high on their priority list ATM.
WRT the limit switch problem, I think there have been some changes to state logic between master and edge but I didn't study it in depth, or if I did, I don't remember the details.
When you change to a different version of g2core, it is a good idea to check the cncjs console for error reports during the initial startup sequence.  Sometimes g2core changes the names of certain json properties and then when cncjs issues a configuration command, g2core chokes on the names it no longer recognizes.  g2core does a poor job of backwards compatibility.

karoria, 2019-04-13:
Thanks for the hint Mitch..

closed 2019-06-05 by MitchBradley