cncjs/cncjs-shopfloor-tablet #14: Plain English

waimate, 2019-04-30:
Obviously I have never used a Pi before. I regret getting it now as I find the process of operating it to be utterly infuriating. What an absurd way to do things. All this mind boggling code. Why can't I just click and drag like on my Mac. Less than one second to install a program. I've been trying for weeks and have got nothing but more agitated!

O.K. so i have followed the instructions and may or may not have installed cncjs on my Pi. I wouldn't have a clue because I can't find it and i can find no instructions on how to go about doing that or running it.
I understand it has to be run through the browser. I can find that because there's an icon to click. 
See, they do know about clicking an icon, but I have to go through this terminal crap to do anything. ridiculous!

So now I realise I have to install the shopfloor tablet version and i have found the unintelligible instructions. I'm fine with downloading the .zip file and I can probably find out within a week how to create this directory thing, but you lost me here.
"Tell cnc to use that directory as the "watch directory" by adding a line like this to the .cncrc file, after the opening brace:"
Who for God's sake is cnc?
What on earth is an opening brace?

And now we've got
 "   "watchDirectory": "/home/pi/GCode",
Use cnc's -m option to set up a static mount. Assuming that the files are in the directory /home/pi/cncjs-shopfloor-tablet, the command would be:

$ cnc -m /tablet:/home/pi/cncjs-shopfloor-tablet/src"

Watch directory? What on earth is that?
Now we've got cnc with an m-option!
For god's sake, what is this nonsense?

If I wasn't so fixed into using this Pi I would throw it away, but I have built my machine with space only for it and now I'm sunk.

Does anyone here speak ENGLISH??

MitchBradley, 2019-04-30:
Who would have thought that a $30 computer designed for DIY hobbyists by a non-profit charity, with the express purpose of allowing people to experiment and modfy, isn't as fast or as easy to use as a $1000 computer designed by a trillion-dollary company, who makes products that are intentionally difficult to even repair?  The mind boggles.

What you don't seem to understand is that everything about cncjs is done by volunteers who have no tangible incentive to do anything. I didn't have to publish cncjs-shopfloor-tablet and I don't have to spend any time at all helping the two or three people who seem to be using it - but I do.  I probably should spend that time doing something else instead.

Regarding your claim that the documentation is not written in ENGLISH, suppose that you were describing an auto repair procedure for an audience that you assumed to be car tinkerers, and you used the word "carburetor".  Then somebody who admits to knowing nothing about the topic comes along and blasts you for not explaining every word that you used.

If you have ever written any documentation for anything (I get the impression that you have not, or you would already get this point), you know that there is a deep tradeoff between getting to the point and diving deep into every little detail.  If you dive deep, there are risks that you will give up for lack of time before you finish (and thus not accomplish the goal of getting to the point), and that the result will so long that you will lose the reader.

I fully agree with your point that much about cncjs is not well-documented, which I why I have personally spent quite a bit of time editing the wiki for clarity and writing introductory material and suggesting nomenclature changes to make it easier to describe.  This is an ongoing process that is far from complete.  Documenting a system the size of cncjs is hard.  Really, really hard. And really, really time-consuming.  My daughter gets paid big bucks for writing about commercial software. Those of us who document cncjs get paid zip, nil, nada for our effort.  But we do get blasted for not having done better and not having spent more uncompensated time on it.

Okay, now that we have gotten our rants mostly out of the way, I will answer your questions.

1. Rant about difficulty of installing on Pi and no icon to click

First of all you need to know that the main developer of cncjs is not focused on Pi usage but rather on big machines with fast processors and lots of memory.  There are a few of us who run cncjs on Pi, but we all have different setups, so an easy-to-use installer that you can run as a no-brainer and it just creates an icon that you can click on would have an audience of maybe two people.  And it would have to be updated frequently as cncjs evolves.  And the other ten people who use cncjs on Pi would complain that it didn't work for them (because their setup is different).

There are in fact easy-to-use, click-on-an-icon, versions of cncjs for Mac and PC, but for various reasons such as the differences between $30 computers and big computers and the hobbyist focus of the Pi software ecosystem and the many different distros (sorry, 'distro' is computer jargon, not plain ENGLISH) that run on Pi, doing one for Pi is not so easy.

You do NOT have to use the shopfloor tablet version.  The full cncjs user interface will run on Pi, using Chromium as the browser.  I personally did not find it suitable for my particular use case, which is controlling a milling machine in a small factory, so I developed cncjs-shopfloor-tablet to provide a more streamlined control surface for my own needs.

To use the full cncjs user interface on a Pi, you start a browser (Chromium seems to work okay, but I am not sure about the  browsers that are used by default on Pi Linux distributions) and browse to http://127.0.0.1:8000 .  That should be written down somewhere, but it might not be easy to find.  If you want an easy-to-use, well-documented, supported CNC solution, you might consider buying one like Mach 4 or UCCNC.  The money you spend goes to paying the people who make it easy to use and document and support it.

2. Question about "what is cnc" with lots of attitude throw in to make people happy to help you

A CNCjs system has several parts, as described [here 
](https://github.com/cncjs/cncjs/wiki/Introduction).  The server component is a program that, prior to a recent change, was named "cnc", but the new name is "cncjs".  There is a todo item for all of us to dig through the documentation and make the name change.

The Raspberry Pi setup guide talks about using 'pm2 start'.  The purpose of that is to cause the server component (now named 'cncjs', formerly 'cnc') to run automatically.  Then you can talk to it with a browser, which can run on any machine on the network.

3. -m option

The main cncjs user interface is built-in.  If you start the cncjs server application (typically via 'pm2 start') without any -m options on the command line, the only UI (that's computer jargon for "user interface", not plain ENGLISH) that is available is the main one.  If you want to use an alternative simplified UI like cncjs-pendant-tinyweb or cncjs-shopfloor-tablet, you must tell the server to add it in.  The way you do that is by adding something to the command line.  If you add '-m /tinyweb:/home/pi/tinyweb', that tells the server that you want the additional tinyweb UI, whose code lives in the directory /home/pi/tinyweb.  The way y [... cut ...]
5. Opening brace

"Opening Brace" is one common name for the character '{'.  See  https://en.wikipedia.org/wiki/Bracket.

The syntax of the .cncrc configuration file requires that the first character be '{', i.e. an opening brace.  The watchDirectory line can go right after that.  There are other places it can go, but it is hard to explain in ENGLISH if you do not understand the Javascript language.

If you have other questions, you might consider apologizing, and then I might consider answering them.

waimate, 2019-05-05:
>    1. Question about "what is cnc" with lots of attitude throw in to make
>    1. -m option
>    1. watch directory
>    1. Opening brace

Well, that is enlightening!
I thought there might be a faceless corporation behind that page, not a
real person of real merit.
I vented my frustration where I could without thought of consequence.
Better than cursing to myself I thought: Not so!

I had no intention of causing upset and I whole heartedly and unreservedly
apologise for that.

My comments and my frustration are both born of ignorance.
I had no idea how small and personal this operation is.  I swam in waters
too deep and have now withdrawn from my
ill-conceived pursuit.

I thank you for the information you have provided which has aided my
understanding of the concept of the Pi.
I thank you for providing software to run a CNC machine.

Regards,

Stuart Pearson.

MitchBradley, 2019-05-05:
Apology accepted.  If you have additional questions, just ask politely and a few of us will be glad to answer in our spare time. If you have suggestions for specific improvements to the wiki, please supply additional text or convert answers to questions into coherent explanation.

closed 2019-06-05 by MitchBradley