cncjs/cncjs-shopfloor-tablet #15: Open Gcode File

qiwalker, 2019-05-08:
Hi Mr Mitch

how to open gcode file using shopfloor 
i cannot get list of gcode file

need your guidance again

thanks a lot
Rizki

MitchBradley, 2019-05-08:
On the computer that is running the cncjs server code, make a directory that will contain gcode programs.  For example, if you are using a Raspberry Pi, that directory might be "/home/pi/GCode".  Then, inside the configuration file ".cncrc" in your home directory, add this line:

   "watchDirectory": "/home/pi/GCode",

Then you can restart the server program and reload the shopfloor web page, and any files in that directory will appear in the dropdown file list.

qiwalker, 2019-05-13:
Thx mr.mitch you are the best

Rgds
Rizji

closed 2019-05-13 by qiwalker