cncjs/cncjs-shopfloor-tablet #16: Support for keyboard shortcuts

karoria, 2019-05-11:
Hi @MitchBradley 
Hope you are doing good. I have developed a shortcut keyboard using "pro micro" arduino compatible board which has native HID capability. This keyboard works great with the main CNCjs web client. For some reasons, it is not working with Shopfloor Pendant. So I can understand that such support is not embedded in shopfloor pendant at present.
I request, if possible for you to enable/support all the keyboard shortcuts of CNCjs so that by making an MOP under the shopfloor tablet touchscreen UI in my main machine panel just get more user friendly. Waiting for your comments/answer.

MitchBradley, 2019-05-12:
I just pushed a [commit] (https://github.com/cncjs/cncjs-shopfloor-tablet/commit/60a059745d090d3e3cf5c77f134410af5901beed) that handles arrow keys for XY jogging, Page Up/Down for Z jogging, and Pause or Escape to pause the workflow.

Supporting all the CNCjs shortcuts would be a tall order because I would have to figure out what they are and figure out how to map them into the limited functionality of shopfloor-tablet.  If you want to add more yourself, feel free.  The code is at the very end of app.js, and it should be easy to understand.  You can find the names of the screen buttons by looking in index.html.

karoria, 2019-05-12:
Thanks a lot for that. I will do my homework for the remaining buttons.

karoria, 2019-05-13:
Hi @MitchBradley,
1. How to write code for combination keys? Can you give an example for Homing? For main CNCjs its shortcut is Ctrl+Alt+Command+H
2. How to write keyboard code for the functions not available in shopfloor tablet? e.g. move 10 times of jog value is Shift+Arrow
Thanks in anticipation.

MitchBradley, 2019-05-13:
I pushed a commit to do the Shift + Arrow (*10) and Ctrl + Arrow (/10).  Also you can cycle through the jog increments with + (or = which is unshifted +) and -.

I am not thrilled by Ctrl-Alt-Command-H.  Homing is typically done infrequently and typing that goofy chord seems like more trouble than selecting from the menu.  If you really want that you will have to add it yourself.  The code to invoke homing is `cnc.controller.command('homing')` which can be seen as the "onclick" handler for the Homing item in the dropdown menu in index.html.

karoria, 2019-05-13:
Sir, you have done all the necessary part of the shortcut keys. Thanks for being so active and helpful. My concern for Homing shortcut is just that I have defined only Single hard button which can emulate that key combination and I want to keep the same button working for both user interfaces (full and shopfloor). I appreciate your code for homing but it will not work with main cncjs interface if I assign hard button to your code. I need further guidance on this.

MitchBradley, 2019-05-13:
You could just add an entry for "h" that calls cnc.controller.command('homing') .  The program will ignore the other modifier keys and run the the homing command when the "h" comes through.

karoria, 2019-05-13:
Ok. Got it. Thanks a lot.

birthstar, 2019-05-24:
Hi just browsing code and think the following might be a bug in app.js for the keyboard shortcuts. 

	case "ArrowDown":
	    jogClick('.jog-z-minus');
	    break;

Probably should read 

	case "PageDown":
	    jogClick('.jog-z-minus');
	    break;

Cheers

MitchBradley, 2019-05-24:
Thanks.. Fixed by

https://github.com/cncjs/cncjs-shopfloor-tablet/commit/236f3573f9cf07a6e409d5e8069211e6b1091c4b

closed 2019-05-24 by MitchBradley

karoria, 2019-06-03:
Hi @MitchBradley 
Thanks for keyboard support in shopfloor tablet. After some testing, got 2 points for improvement as under:
1. When jogging by arrow keys, if we press the same arrow key 3 or more times immediately, the axis malfunctions and goes in opposite direction of where it suppose to go.
2. You have put Ctrl+Arrow for 0.1x jog, where in main cncjs console it is Alt+Arrow. I would recommend to use same key combinations because when using both the interfaces, user will forget which keys to use.
Regards,
Ravi

MitchBradley, 2019-06-03:
I changed Ctrl to Alt per your request.  Pull the latest code.

The other problem is less clear.  On my setup, 3 rapid-fire presses cause no problems at all - but sometimes if I do, say, 10 in a row I start to see GCode errors reported in the console window of a side-by-side full CNCjs UI.  I "fixed" it by removing some delays at the end of jog sequences.  I suspect that GRBL was losing input while it was waiting.  I don't know if this fix will work for you or not.  It is possible that your special keyboard does not like to send rapidly.

karoria, 2019-06-04:
Thanks @MitchBradley
I will check and update you. Further inputs:
1. I was using wireless keyboard to raspi so I replaced it with wired usb keyboard but the problem remains same.
2. My keyboard is attached with server. I will  check it with client also to get some clue.

karoria, 2019-06-04:
@MitchBradley 
OK. I have tested new code and the problem remains same for 3 or more immediate presses of arrow keys. The problem remains with server as well as client. Main cncjs UI works well with keyboard. I suspect that g90 and g91 conversion (in shopfloor tablet) while jogging makes this problem, as I could see that it returns mostly to 1.000 location when jog increment is 1.000 regardless of its position. I mean if this happens at location 5.0 then also it returns to 1.0 and if its location is 55.0 then also it returns to 1.0 while pressing keys multiple times.
I hope I could explain this well.

MitchBradley, 2019-06-04:
I agree with your idea that the problem is related to G91.  The rapid-fire commands are probably causing the controller to miss a G91 so the next G0 X1 is interpreted in absolute mode.  However, there is no obvious difference between shopfloor-tablet and full cncjs UI in this regard.  In both cases, the jog sequence is the same:  G91<newline>G0 X1<newline>G90<newline>.
You can verify this by looking in the cncjs Console window; you should see:
```
feeder> G91
feeder>G0 X1
feeder> G90
ok
ok
ok
```
for both UIs.  In some cases the first ok will happen in the middle of the sequence, again on both UIs, as a result of the exact timing of the GRBL interpreter.
There was a version of CNCjs in which the jog sequence had only two lines:
```
G91 G0 X1
G90
```
It was changed to the three-line version when Marlin support was added, because, contrary to GCode standards, Marlin does not permit G91 and G0 on the same line.  Marlin is troublesome in many other ways as well...
Final thought: Make sure you have the absolute latest version of shopfloor tablet - commit https://github.com/cncjs/cncjs-shopfloor-tablet/commit/d22757a4360e8642ea7b355f790d2f4f25040ec2 .  It removes a post-jog delay that caused problems for me.

karoria, 2019-06-05:
Ok. I will check and update you. I am using g2core.

MitchBradley, 2019-06-05:
Ah - I wish you had told me that earlier.  I do experience the problem with g2core.  I think I know what is causing the problem.  There was a bug whereby, if you used MDI to issue a G91 to go to relative mode, then a jog command would cancel that.  I think there is now a synchronization problem whereby the information about the current mode is not valid when the next jog is issued.  This kind of synchronization issue is endemic to the ecosystem of tiny micros running motion control, fed over serial lines from bigger computers, with multiple levels of queues in between.  I have been spending inordinate amounts of time trying to improve the situation, but it is a really hard problem.  I am essentially trying to make a professional quality system out of components that evolved from garage projects.

MitchBradley, 2019-06-05:
Yep, that's the problem.  The first jog command sends G91 for relative mode, then G0, then G90 for absolute mode.  When g2core processes the G91, it sends a report that the system in in relative mode, but before it runs the G0 and G90 and reports absolute mode, another keystroke comes in, so the next jog is sent as a G0 without the preceding G91.  Maybe I can ignore mode reports while jogging...

MitchBradley, 2019-06-05:
I think I fixed it per the referenced commit.

karoria, 2019-06-06:
Hats off to you. So quick!! I will check new commit and update you.

karoria, 2019-06-06:
I have tested your new code and works like a charm in my machine. Thanks a lot!! One thing I noticed when I am jogging with long press on any key and when axis reaches the soft limit, then machine behaves the same way (temporarily goes to g91, so brings the axis back to 1 or jog increment) this is also seen on the screen that g90 changes to g91. This I noticed when my z gone to top and my finger was still on page up and OMG! it came down with rapid and I struggled pressing emergency in that short time. But everything gone OK!!
Rest, I would say the movement of axis by long pressing the arrow keys is butter smooth, even better than full UI. You have done a great job. I love to play with mini wireless keyboard and now with confidence I can move the axis to the point with very short time.
I request to find out the cause of above problem and its solution at your time.
Thank you very much once again.

MitchBradley, 2019-06-06:
I cannot reproduce this.  When I set soft limits and hold down an arrow key, it stops at the soft limit.

If you send me logs from the full UI console window, I might be able to discover the problem.

Alternatively, you might be able to stop the problem by changing this code in app.js:cnc.sendMove from

	if (modal.distance == 'G90') {
            controller.command('gcode', 'G91'); // relative distance
            controller.command('gcode', 'G0 ' + s);
            controller.command('gcode', 'G90'); // absolute distance

to

	if (modal.distance == 'G90') {
            controller.command('gcode', 'G91 G0 ' + s);
            controller.command('gcode', 'G90'); // absolute distance

The result would not work on Marlin but you probably don't care about that.

karoria, 2019-06-06:
Sure. I will give it a try. Thanks. Let me ask you for soft limits. In g2core master branch, reaching a soft limit issued alarm and you need to unlock it before you jog back. Now with edge branch, i dont see alarm upon reaching soft limit. It just stop jogging in that direction but can be reversed without unlocking alarm. Is it an improvement? Or i am missing something to get it right?

MitchBradley, 2019-06-06:
Whether or not it is an improvement is a matter of taste.

MitchBradley, 2019-06-07:
Even safer: change:
```
	if (modal.distance == 'G90') {
            controller.command('gcode', 'G91'); // relative distance
            controller.command('gcode', 'G0 ' + s);
            controller.command('gcode', 'G90'); // absolute distance
	} else {
            controller.command('gcode', 'G0 ' + s);
	}
```
to:
```
        controller.command('gcode', 'G91 G0 ' + s);  // Always force relative distance
	if (modal.distance == 'G90') {
            controller.command('gcode', 'G90 ');  // Back to absolute
	}

karoria, 2019-06-07:
Ok. Got it. Will try and revert. Thanks.

karoria, 2019-06-08:
I have tested with both the solution but still no success. First solution didn't change anything as if I haven't changed the code, while the latest one couldn't do keyboard jogging at all.
Further input:
1. This happens at maximum (soft) limits only (in all x, y and z axis), not the minimum limits (or zero)
2. When I just reach at max soft limit and short press the key any number of times, nothing will happen but a long press at that location (max soft limit) converts the mode to G91 which can be seen on screen too and the axis goes to the absolute coordinate equal to jog increment.
Thanks.
Ravi

MitchBradley, 2019-06-08:
Are you going to send me console logs like I asked?

karoria, 2019-06-08:
Sure. I am out of station now. Will send you probably on Monday or Tuesday.

karoria, 2019-06-14:
Hi @MitchBradley 
Attaching here console log as well as brief video. I've done X+ with 5mm jog distance 3 times and then long pressed the jog arrow key till X maximum soft limit. The axis stays at soft limit for a while and then returns back to 5 absolute position, as jog distance is 5.

[Console Log 140619.txt](https://github.com/cncjs/cncjs-shopfloor-tablet/files/3290122/Console.Log.140619.txt)

[VID_20190614_145429106.mp4.zip](https://github.com/cncjs/cncjs-shopfloor-tablet/files/3290133/VID_20190614_145429106.mp4.zip)

MitchBradley, 2019-06-14:
This works for me:
```
    var jog = function(params) {
        params = params || {};
        var s = _.map(params, function(value, letter) {
            return '' + letter + value;
        }).join(' ');
        controller.command('gcode', 'G91 G0 ' + s);  // Always force relative distance
        controller.command('gcode', 'G90 ');  // Back to absolute
    };
```
It does not respect a previous setting of G91, instead always returning to G90 mode after a jog, and it won't work on Marlin, but in your setup it will probably be an improvement.

karoria, 2019-06-15:
Didn't work for me.. :(

karoria, 2019-06-15:
Just to reassure you that this happens when jogging with keyboard only. The UI jogging works ok all the time. However long press doesn't work there which is good for touch screen.

MitchBradley, 2019-06-15:
"Didn't work" is virtually useless for debugging.  I need logs.

karoria, 2019-06-19:
Oh sorry @MitchBradley 
I don't know how but this time it worked. I might have missed something when replacing the code that time. Today I tried again and it worked!!
Thanks for your precious time and support.
Best,
Ravi

MitchBradley, 2019-06-19:
Maybe you can find a jog step setting that matches the keyboard auto-repeat rate, so the jogging stops soon after you release the key.

karoria, 2019-06-19:
Telepathy!!! Working on that only..:)