cncjs/cncjs-shopfloor-tablet #18: Feature request: Message/Alarm/Override display

karoria, 2019-12-05:
Hi @MitchBradley ,
I am using shopfloor tablet for almost a year now intermittently and would like to say it really does its job beautifully. I run it in my 7 inch and 10 inch raspberry pi displays as machine hmi and it functions well. The only time I need to go to main cncjs is for ALARMS, MESSAGES (like tool change) and to see OVERRIDES.
Right now I use grbl with ESP32 and I would like to see those messages/alarms at the bottom of the screen so that user doesn't need to go anywhere else. I can remember that you added override buttons in one of your versions and due to inability of g2 for overrides, you withdrew those buttons. I request if you can bring them back only for display purposes if not controllable or at least for grbl which has robust override control. BTW, I am creating knob for different overrides by programming pro micro as hid keyboard. So I can change overrides from there but can't see it on display.
Waiting for your reply..
Best regards,
Ravi

MitchBradley, 2019-12-05:
I will look into this.  I have already done some work on reporting messages, so perhaps I can get something along those lines quickly.

karoria, 2019-12-05:
That's great news. Eagerly waiting for it...

karoria, 2020-02-19:
Sir, I can't resist myself writing here to know the status despite of knowing you run busy with multiple projects and this task may not be in your priority list. Your two quick words will help me. Also I want to know if I am using only 1 controller (say grbl), how to set it permanently with shopfloor tablet without showing other options to user. Thanks in anticipation.

samgonemudding, 2020-04-29:
Wow, just getting started with cncjs tablet and it is a PERFECT adjunct. Love it. Just wanted to add that with tool changes, the deskdop requires a "resume" command that is distinct from the "run" command to continue. If there is a way to add/modify the start button to perform "resume" function, that would be pretty great. Thank you. Amazing contribution.

karoria, 2020-04-29:
The start, pause, resume and stop commands are dynamic in shopfloor tablet. It means the buttons change their role as per need. You will see, when you run a program you will see pause button. After making a pause (feedhold) when you press start again, it will act as resume. Do this first by dry running a program and get used to these commands, then only go for the real jobs.

MitchBradley, 2020-04-29:
The logic behind the way it works is this:
When you are running a job and need to stop quickly, you do not want to have to think about which button to hit.  There is only one active button - Pause - which will make the machine stop.  Once the machine has safely stopped, you can then choose whether to resume from that point (green) or stop the program entirely.
If  you want to re-run the program from the beginning, you must stop and then hit run.  That is safer and less prone to human error than having Resume and (Re)Run buttons active at the same time.
I tuned this user interface behavior based on my experiences - and mistakes - in the workshop, while running production jobs.

karoria, 2020-04-30:
Hi @MitchBradley 
I just wanted to update you that I have modified some of the interface of shopfloor tablet with my limited knowledge and got overrides working with realtime display of speed (like spindle rpm after override applied) as well as added keyborad shortcuts for easy operating. I have just done this for grbl. I know this is a small thing for you but if you want to have a look, please visit my fork. Btw, I am not familier with pull request, commits etc.

samgonemudding, 2020-04-30:
> The logic behind the way it works is this:
> When you are running a job and need to stop quickly, you do not want to have to think about which button to hit. There is only one active button - Pause - [...]

Those are very excellent features and I gathered as much. Great design overall - very functional. My issue was a little different... I just use a simple Shapeoko 3 for my stuff and Carbide create drops an M6 command at the start no matter what. Probably a good thing but 90 percent of my work is a quarter inch endmill - I don't do a lot of tool changes. When I hit start, CNCJS sees the M6 command and asks if I want to proceed. I can continue by hitting the resume button on the desktop, but hitting the tablet only shows "start" and I don't see a "resume" call in my brief sweep of the Javascript. Either way, I have to hit resume on desktop everytime, which is probably a good safety catch. For now I am just deleting the M6 command in the cgode. 

Thanks for the quick reply guys!

MitchBradley, 2020-04-30:
Post a link to your fork and I will have a look at it.

karoria, 2020-05-01:
https://github.com/karoria/cncjs-shopfloor-tablet

karoria, 2020-05-05:
> Post a link to your fork and I will have a look at it.

Sir, have you seen my changes? Any comments for me? I have still "Alarms" and "Messages" display is pending. I am learning javascript now during this lockdown.

MitchBradley, 2020-05-05:
I did look at it.  It has many changes that are specific to your setup.  One rule to make it easy for people to accept your changes is to remove unnecessary things so the upstream people, who are usually very busy, can decide on small, targeted changes.  So my advice would be to create a branch that has only the new features, with no changes that are only for you. Then create a Pull Request by clicking on the Pull Request button.  Pull Requests make it easy for upstream people to handle the new code.  They can easily comment on specific lines, request changes, and accept changes with a few clicks.

karoria, 2020-05-05:
@MitchBradley Thanks for your valuable comments. Yes, I have made changes which are specific to my machine which runs on grbl with metric units. I might have deleted settings/code which pertain to g2core, smoothie or marlin. If you get time, run that code with your grbl machine. It works really nice with so many keyboard shortcuts added. I use a wireless keyboard and thanks to your last code update ragarding "when a text entry field is activated, the shortcuts don't work". This has made it so easy for me to add keyboard shortcuts with single key.
Once I complete javascript learning, I will surely work on your suggestions and try to make Pull Requests (which are not specific to my machine) and hence can contribute to open source community.