cncjs/cncjs-shopfloor-tablet #19: Connection issue when using larger files

karoria, 2019-12-26:
Hi @MitchBradley 
When i already have connected controller with main cncjs interface and I open and connect tablet in another tab, the system hangs. Particularly this happens when i use programs larger than 6MB. I have also noticed that shopfloor tablet hangs while loading larger files individually. (Main interface not connected).
What can be the possible reason? I use RPi4 with 2GB RAM. Router is connected to RPi with ethernet and I use local RPi display and chromium browser. Let me tell you, initially I used wifi and then switched to ethernet. By doing this, it started responding faster for smaller files.
Another observation is, some programs of approx 5 to 6MB sizes are opening smoothly in main interface while it doesn't load in shopfloor tablet and make system unresponsive.

karoria, 2019-12-26:
Sorry, forgot to mention that I am using grbl with ESP32 controller.

karoria, 2019-12-28:
@MitchBradley 
Any clues?

MitchBradley, 2019-12-28:
Busy with other stuff

karoria, 2019-12-28:
Ok.