cncjs/cncjs-shopfloor-tablet #22: [Support] Y axis reversed

Billiam, 2020-04-05:
I'm running CNCJS with a grbl CNC router, where 0,0 is in the back rear corner when facing the machine.

The UI for shopfloor tablet shows my Y-axis reversed, as are the arrow keys when using a laptop. Directions and coordinates are correct in the normal CNCJS interface. 

Is there a setting to change this and other coordinate issues that may be affected?

MitchBradley, 2020-04-05:
shopfloor tablet assumes that X increases toward the right and Y toward the back.  Knowing the 0,0 point does not reveal the direction of the axes.  To understand this better I would need to know the way your controller (Grbl, etc) is configured, how limits are set up, etc.

Billiam, 2020-04-05:
> X increases toward the right and Y toward the back

This is the case for my machine.

I think the surprising behavior is that the tablet UI though shows 'Y+' in the downward direction  [image].

My machine moves correctly, in that Y+ moves the gantry toward the rear of the machine, but I would expect that to be the topmost button in the UI and the up arrow on the keyboard, instead (matching CNCJS).

Billiam, 2020-04-08:
Amazing. I'm using the wrong fork. Everything is fine, forget I said anything 😆

closed 2020-04-08 by Billiam