cncjs/cncjs-shopfloor-tablet #26: [Feature Request] Need to be able to run Macros

wb9tpg, 2020-10-11:
I have a number of Macros I use for a multiple station complicated job.   It would be extremely helpful to be able to run macros directly from shop floor tablet.

pedwards71, 2021-11-02:
I'd love to be able to run macros too. I would use it for tool changing, unless there's already a way to do that from this interface?

I run this interface on an iPad at the machine. It's very useful. Had a few bad things happen when using the main cncjs interface from a touch screen, because of accidentally touching the wrong controls - the exact reasons that this interface was created.

Thanks for the great work!

MitchBradley, 2021-11-02:
You could just put the macro into a named file and run that.  There is very little difference between a GCode file and a macro.

MitchBradley, 2021-11-02:
I guess there is the variable substitution thing.

MitchBradley, 2021-11-02:
Actually, now that look inside the CNCjs server code, I think that variables work the same way regardless of whether you are running a macro or a GCode file.

MitchBradley, 2021-11-02:
The way I organize my jobs is to have a folder for each production job.  Files in the folder are named like "0Setup.nc", "1Top.nc", "2Roundover.nc", etc.  I just run the files in numerical order, doing whatever manual steps are necessary between steps.

pedwards71, 2021-11-03:
Ok. So instead of having one nc file that does the whole job with tool change commands inserted, you have separate files for each different tool section and have a separate file for doing tool changes?

Sounds ok. I'll give it a try. Thanks for the tip!

BTW, I usually generate my nc files from FreeCAD. I like to generate one file with all the tool changes, which works best for simulation, but it wouldn't be hard to split it up to feed it to my machine, which is a Shapeoko 3 XXL.