cncjs/cncjs-shopfloor-tablet #35: Z-number is not working

Smeierstad, 2022-03-29:
Just installed the shopfloor pendant, and noticed that the z-number on the shopfloor pendant is moving with the y-number. If I move the y-axis 10mm, it says on the UI also has moved the z-axis the same amount. The z-axis does not move on the machine, just the on the UI. I can also control the z-axis with the UI, but the number will not change. This is only happening on the shopfloor UI, I have tested it on the regular cncjs UI.

MitchBradley, 2022-03-29:
I do not see any obvious problems in the source code that could cause this problem.  The Y and Z displays work for me.  How did you install the shopfloor pendant?

Smeierstad, 2022-03-30:
I installed it with the cncjs-pi-raspbian installer.

MitchBradley, 2022-03-30:
I have no idea which version of the cncjs-shopfloor-tablet code is in that installer.  In the past I have had problems with installer packages that have ancient versions of code.  Install using the instructions in the README at https://github.com/cncjs/cncjs-shopfloor-tablet .  I cannot and will not support old versions.

closed 2022-03-31 by MitchBradley