cncjs/cncjs-shopfloor-tablet #37: Problems with incognito mode

wreutz, 2022-10-07:
Hi,
I'm using the install script for cncjs and activated the kiosk mode. This works but I have problem:

I use the url http://localhost:80/tablet to open in tablet view but it always falls back to the default web interface.
Tried it on my browser and if I open a private browsing window or opening a fresh browser profile directly with http://<IP address of raspberry pi>:80/tablet I get redirected to the default interface.

when I take a regular window or if I disable incognito mode on the raspberry pi kiosk it happens as followed:
- first boot -> redirect to default interface
- all other boots -> tablet interface shows

so it must have something to do with cookies / local storage etc.

Would it be possible to fix it that the redirect is omitted when directly accessed for the first time like in an private browsing environment?

mrmazakblu, 2024-05-03:
i see the same or at least very similar situation. The first time to open browser, then go to /tablet, i am forwarded to full interface. Second time open /tablet i  get tablet UI.

Difference is I am not using incognito or kiosk mode.

MitchBradley, 2024-05-03:
I don't know how to fix it.  I don't understand all the details of how the redirect works.  If anyone can supply a patch, I will apply it.