cncjs/cncjs-shopfloor-tablet #38: The displayed Gcode in the toolPath canvas appears to be squashed in the Y axis

ElectricBluefishProductionsInc, 2024-12-12:
Love the Pendent, however, I may have found a bug?
Should display a near perfect circle, but the display appears to be compressed in the Y axis.
[image]
 The values displayed on the output are **1365.59mm**, **1363.04mm**, 0.00mm, 0.00mm.

MitchBradley, 2024-12-13:
You may indeed have found a bug, but without knowing the gcode that resulted in that display, there is no way to tell.

MitchBradley, 2024-12-13:
Another bit of information I would need is details of how you are displaying the pendant.  It was designed for fullscreen display on a tablet with a fixed screen size.  If you are displaying it in a resizable window on a PC, it is possible that a resize operation would change the resolution and aspect ratio of the visualizer window without recomputing the display.  As with any problem report, solving the problem without knowing almost everything about the context is almost impossible.